LibreSolar/bms-8s50-ic #3: Add free running diodes parallel to MOSFETs

xsider, 2019-08-14:
Two shottky diodes should be added, parallel to the body diodes inside the MOSFETs. We had maybe a overheating event occured at Switch-N-Sense when huge amount of currenct was sink from the battery while the charge mosfet was open, so the body diode was used. The body diode is a bad one (big voltage drop).

martinjaeger, 2019-08-14:
This can be solved in firmware with improved control (ideal diode control). Monitor the current and as soon as it goes into discharging direction switch on the charge MOSFET even if charging is not allowed at that moment.

martinjaeger, 2019-08-18:
Just saw that the ISL94202 supports ideal diode control out of the box. Look at the DFODUV and CFODOV settings.

martinjaeger, 2019-08-18:
Closing this issue, as solutions without hardware change exist. Please reopen if you disagree.

closed 2019-08-18 by martinjaeger

martinjaeger, 2021-06-24:
FYI: I have implemented the ideal diode control today.

https://github.com/LibreSolar/bms-firmware/commit/42f2da214415df28c67f6d3dba1b5db2b75b85b2