LibreSolar/bms-8s50-ic #5: voltage measurement of cell 1

MatthiasN73, 2020-02-05:
Hi,
I recently finished building the bms-8s50 and encountered a problem with the voltage measurement of cell 1. The measurement is very instable. the voltage oscillates by approximately 200 mV and is about 500 mV to low. After some investigations and bridging R59 the oscillation is gone and near the real value but still to low (150 mV). What is the function of R59?
By the way I studied the datasheet of the ISL94202 and found some issues: 
a missing capacitor between VC0 and Ground (47nF)
a missing resistor (1M) between source and gate of Q10 (PCFET)
wrong value of C30 should be a 470nF capacitor
If I have understood the datasheet correctly, the Precharge path should bridge the charge fet in case of to low voltage of one or more cells. But in your case the discharge fet is bridged. Why?

sorry for my english.

martinjaeger, 2020-02-06:
Hi Matthias,
very cool that you rebuilt the BMS. Unfortunately I didn't have much time for it in the past months, so there was not too much progress in testing and firmware of the BMS.

I didn't have any issues with cell voltage measurements in my 4s battery pack so far. But I'll double-check.

The idea for R59 was to decouple the power path from the analog measurement ground. I actually took this idea from the [DieBieMS](https://github.com/DieBieEngineering/DieBieMS). 

You are right, a capacitor between VC0 and GND is missing. This might cause some noise in cell 1 measurement, but it should not result in 150 mV offset... but it's definitely an error that should be corrected in next revision. It could be possible to add the capacitor manually attached to R25 in current PCB layout.

Also the value of C30 is wrong, I agree.

I'm using the precharge circuit different from what is stated in the datasheet. I use it to precharge the capacitors of devices connected to the battery before switching the discharge MOSFET on. This prevents high inrush currents. I don't really see the point of precharging the battery (not the capacitors) slowly via resistors, as the BMS could just tell the charger to use low current.

Are you using the new Zephyr-based firmware or did you develop something else yourself?

MatthiasN73, 2020-02-06:
Hello Martin,
Thanks for your quick response.
In my configuration, I use all 8 cells.
I would like to use the zephyr version, but there is no board definition for the STM32F072 mcu and so I cannot flash the compiled firmware. 
To get the bms working, I just completed the missing function “isl94202_write_bytes” and “isl94202_read_bytes” in the mbed-version.

martinjaeger, 2020-02-06:
As I'm now quite used to Zephyr, it shouldn't be that much effort to add a board definition for the STM32F072 MCU. I'll try to include that.

However, as I soldered the L4 MCU onto my board, I can't test it for the new BMS.

martinjaeger, 2020-02-06:
Please have a try: https://github.com/LibreSolar/bms-firmware/commit/85969f8416860dc9ba5ee1d79ab26c3384f68910

MatthiasN73, 2020-02-10:
Unfortunately I can’t compile the firmware. Seems to be too big. 
The following Errors appear:
firmware-pre.elf section `bss' will not fit in region `SRAM'
region `SRAM' overflowed by 13320 bytes
Any idea?

martinjaeger, 2020-02-10:
Oh yes, the heap size was way too high (which is not necessary). Reduced it to 1024 bytes. Please `git pull` and try again.

Here is the change: https://github.com/LibreSolar/bms-firmware/commit/0e6d23e10f0c196ae55d72aa18770a512bd60185

MatthiasN73, 2020-02-10:
Now build and upload works. Thanks again.

MatthiasN73, 2020-02-10:
This is an automatic reply to your message
 
 ---------------------------------------------------------------------
 Sie haben eine E-Mail an die Adresse m.neubauer@schenk-ansorge.de geschickt.
 Leider bin ich bis einschließlich 14.02.2020 18:00  nicht im Labor. 
 In dringenden Fällen wenden Sie sich bitte an Herrn Gerhard unter
 der Telefonnummer 0391/24468-246 bzw. F.Gerhard@schenk-ansorge.de

 Vielen Dank für Ihr Verständnis.

 ---------------------------------------------------------------------

 File names:            
 
Oh yes, the heap size was way too high (which is not necessary). Reduced it to 1024 bytes. Please `git pull` and try again.

Here is the change: https://github.com/LibreSolar/bms-firmware/commit/0e6d23e10f0c196ae55d72aa18770a512bd60185

MatthiasN73, 2020-04-19:
Hi Martin,

after further investigation and reading then ISL94202 datasheet, i discovered the potential cause of this issue.
The VDD Pin of the ISL94202 should conntected to the top of the pack through a low pass filter consisting of a <10 Ohm resistor and a 4.7μF capacitor.
In your circuit VDD is connected through the VBATT low pass filter with an resistor of 100 Ohm.
Currently i've connected D2 through a separat low pass filter (4,7 Ohm / 4.7μF) to BAT+. Now the volatage measurment is okay. 
By the way i replaced C39 with a 1µF capacitor, C30 with a 470nF capacitor and add the missing 47nF capacitor at VC0.

Furthermore i increased the shunt resistor to 0.3 mOhm, but the displayed current does'nt match the measured value of my multimeter.
Any ideas?

martinjaeger, 2020-04-23:
Hi Matthias,

just checked with my BMS here and realized that the current measurement is pretty accurate above 1A, but more than 20% off at very low currents.

The reason may be that the firmware doesn't adjust the current sense amplfier gain yet, it is always kept at the default setting (gain 50). The accuracy will probably become better if it is set to 500.

Great that you found the reason for the issue with VBATT measurement. Didn't have time to test it yet, but will try to do that soon, together with fixing the capacitor values. Afterwards I will also update board and schematic :)

martinjaeger, 2020-05-08:
Soldered all the updates you suggested above to my board. Looks a bit ugly now :)

Also the KiCad files should now include all the identified fixes. Please have a look if I forgot something.

I hoped that the added capacitors would also solve the incorrect pack voltage measurement (see [here](https://github.com/LibreSolar/bms-firmware/blob/8ff63999a65384dbe5ab18886f27a3f042b34043/src/bms_isl94202.c#L290)). Unfortunately, that was not the case here. Did you try out if the pack voltage measurement provided by the chip is correct? I get a pack voltage of around 12.8V, but sum of single cell voltages is 13.1V, which is also what my multimeter tells me.

The shunt resistor is not updated yet (still need to decide which value is suitable, as the current measurement gets inaccurate if the shunt value is too small, as you have seen as well).

MatthiasN73, 2020-05-09:
Hi Martin,

looks good. I replaced the R59 by a 0R resistor, because it affects the measurement as well.
The pack voltage i messured with my multimeter ist 31.0 V (it's a 8S LIPO pack) and the ISL tells a voltage of 31.16.
I think it's pretty okay.

martinjaeger, 2020-05-09:
Ok, that might in fact be the reason. Will try that out aswell and let you know the result.

martinjaeger, 2020-05-11:
Morning! Just had a try with R59 as 0R, but it didn't really improve the pack voltage reading.

Just for clarification: Did you measure above stated values via the ISL94202_VBATT register or with the sum of all single cell voltages (which is how the pack voltage is currently calculated in the firmware).
If I comment out the line referenced above I still get a pack voltage of 12.91V (instead of 13.32V if using the sum of all single cell voltages). The actual voltage measured with the Fluke Multimeter is 13.33V. So the sum of cell voltages is very accurate, while the pack voltage is not at all...

MatthiasN73, 2020-05-11:
Yes, i activated the pack voltage measurement by reading ISL94202_VBATT register.  My multimeter is just a simple PEAKTECH 3335. Maybe it's not very precisely. Will try out the summation of all cells and compare it with the VBATT register value tomorrow.

MatthiasN73, 2020-05-11:
Hi, here are my results:
- sum of cells voltage: 29.68 V
- result of reading VBATT register: 29.79 V
- measurement with multimeter: 29.6 V
I think, this is an acceptable deviation (< 0,5%).